IL2232-HT2022-HIL/SOFTWARE_DEV #12: Update communication protocol

Fkar97, 2022-09-30:
With information from Matthias about how a test can be executed, the communication protocol needs to be updated

sofiecmf, 2022-10-03:
- [ ] Figure out how often to send/receive communication

Fkar97, 2022-10-05:
Hi,

here is a rough sketch of what I thought a test could look like (I thought for the beginning a [keyword driven](http://robotframework.org/RobotDemo/#id5) approach could be simplest). Any suggestions from your side are also welcome.

One of the requirements for the lab could be: After a pedestrian pushes the button at a traffic light it takes up to 10 seconds until the signal turns green. 
```
[Arguments]     ${traffic_light}                                    //argument lets us select which traffic light to test
wait until ${traffic_light} pedestrian_red ON                       //'wait until' an output has a certain state, parameter 1 output name, parameter 2 
output should be ${traffic_light} pedestrian_green OFF              //'output should be' confirms  the state of a given output
push button ${traffic_light}                                        //'push button' sets and resets the button signal 
wait until led ${traffic_light}  pedestrian_green ON timeout 10000  //additional parameter with timeout [ms]
output should be ${traffic_light} pedestrian_red OFF
```

Fkar97, 2022-10-05:
one thing that could be important is the possibility to check multiple outputs at the same time (also wait for one and check others). Since there is some latency between issuing commands and reading the actual state of a pin, in the example above, the green led in line 3 could already be turned on if there was no traffic which would make the test wrongfully fail.

sofiecmf, 2022-10-07:
BNF suggestion for operation of buttons, switches, joystick:

BNF
<test>  ::= ”Set” <task> | ”Set” <task> <test>
<task> ::= <actor> ”to” <state>
<actor> ::= <type><index>
<state> ::= <onoff> | <direction>
<type> ::= ”button” | ”switch” | ”joystick”
<index> ::= ”1” | ”2” | ”3” | ”4”
<onoff> ::= ”on” | ”off”
<direction> ::= ”up” | ”down” | ”left” | ”right”

Example 
Set button2 to on Set joystick1 to right

HolgerStenberg, 2022-11-11:
This is an old task that is now solved

closed 2022-11-11 by HolgerStenberg