IL2232-HT2022-HIL/SOFTWARE_DEV #40: HiL MCU: Implement the potentiometer task

HolgerStenberg, 2022-10-31:
The Shield has a potentiometer on it controlling the brightness of the leds of the shield. Since this potentiometer is a voltage divider, connected to an ADC on the dev mcu board, the HiL board need to emulate this behaviour in software. In order to do so, a DAC (DIGITAL TO ANALOG CONVERTER) output will be used on the HiL-board. 

The task will be to implement the potentiometer emulator using the given DAC output.

**The definition of the task is as follows:**
Input: a value in between 0 and 100, indicating a percentage of the potentiometer
Output: A voltage reflected by the percentage of the input is outputed from the DAC1 interface. 

Suggest a periodicity of the task in order to fulfil the response time for the intended implementation (if not given, make an educated guess).

sofiecmf, 2022-11-10:
- [ ] Create a setup file for dac_start and others

HolgerStenberg, 2022-12-01:
The code is inserted in the Olle_Structure branch. No need for an additional file for setup for now

closed 2022-12-01 by HolgerStenberg