IL2232-HT2022-HIL/SOFTWARE_DEV #49: Python&Document: align python client scripts with its Wiki page

HolgerStenberg, 2022-11-14:
- [ ] align the function names in HiL_client.py
- [ ] check the correctness of description and if the descriptions in wiki match the python scripts

HolgerStenberg, 2022-11-14:
@Yuxiaowa Would you also be able to fill in the documentation for the different client functions in the wiki page [python client function overview?](https://github.com/IL2232-HT2022-HIL/SOFTWARE_DEV/wiki/PYTHON-CLIENT-FUNCTION-OVERVIEW)

Yuxiaowa, 2022-11-14:
@HolgerStenberg Sure.

HolgerStenberg, 2022-11-16:
Hi @Yuxiaowa!
Could you start fill in the documentation for the functions in [this wiki page?](https://github.com/IL2232-HT2022-HIL/SOFTWARE_DEV/wiki/PYTHON-CLIENT-FUNCTION-OVERVIEW). You can also try to see if the naming are all correct. I will keep on working on some decode functionality in the python client while you do this!

Yuxiaowa, 2022-11-16:
@HolgerStenberg Sorry for the delay. I will work on this now.

Yuxiaowa, 2022-11-24:
All of these three were finished.

closed 2023-01-17 by HolgerStenberg